philibertc/micronova_controller #20: Fumetemperatur | Waterpres incorrect

geraldFS, 2021-11-04:
Hello everyone, today was the day. I used the controls in Homeassistant.

Basically it works - but:
- The smoke temperature does not fit a maximum of 10 ° C
- The water pressure is stated three times too high
- Flame power remains at 0 (but you can only see it in the MQTT Explorer

Pellet stove used:
ITC Layma 18kW
Micronova Mod0047 EVO

Addendum to the interference suppression, as soon as I use the 18V from the motherboard, my WLAN quality drops, now I use an interference suppression board (filtering / smoothing the input voltage), the WLAN reception is good.

philibertc, 2021-11-05:
Hello,
To analyze which addresses might match, you can use this program: [test2bis](https://github.com/philibertc/micronova_controller/blob/master/tests/esp8266/test2bis_read_all_addresses.ino)
Which buck converter do you use? I personally have sufficient smoothing with an LM2596.

geraldFS, 2021-11-05:
Hi there,

these are my standard DC / DC converters, they work wonderfully.
But as I said after I repeatedly had problems with other devices, that they received interference from outside, I got used to always smoothing the voltage with capacitors. Since then, all problems have been resolved.

How do I use the test2bis?

I compiled it and uploaded it but then ...

https://de.aliexpress.com/item/4000259403990.html?spm=a2g0o.productlist.0.0.257f1dd3juZuYu&algo_pvid=f9ac536e-0737-48d7-a678-208c7d544981&algo_exp_id=f9ac536e-0737-48d7-a678-208c7d544981-15&pdp_ext_f=%7B%22sku_id%22%3A%2210000001054452173%22%7D

geraldFS, 2021-11-05:
fumesTempAddr = 0x5A

philibertc, 2021-11-05:
> https://de.aliexpress.com/item/4000259403990.html

Ah, okay, these are "mini" models. I'm used to these: https://m.media-amazon.com/images/I/61VvzQgykgL._AC_SY355_.jpg which have a more consistent smoothing.

> I compiled it and uploaded it but then ...

Ah, yes, sorry. You have to open the serial monitor (115200 bauds) to analyze the results and eventually find one corresponding to the value you are looking for.

> fumesTempAddr = 0x5A

It works for you with that value?

geraldFS, 2021-11-05:
```
12:55:41.866 -> 
12:55:41.866 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
12:55:41.900 -> 
12:55:41.900 -> Exception (0):
12:55:41.900 -> epc1=0x43420e04 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
12:55:41.900 -> 
12:55:41.900 -> >>>stack>>>
12:55:41.900 -> 
12:55:41.900 -> ctx: cont
12:55:41.900 -> sp: 3ffffdc0 end: 3fffffc0 offset: 0190
12:55:41.900 -> 3fffff50:  3ffee080 3ffee650 00000046 3ffee5ec  
12:55:41.900 -> 3fffff60:  3fffda00 3ffee650 00000046 402025a4  
12:55:41.900 -> 3fffff70:  40202c65 00000046 3ffee5ec 402010e2  
12:55:41.900 -> 3fffff80:  3fffdad0 00000000 000000cf 40201160  
12:55:41.933 -> 3fffff90:  00000100 00000000 feefeffe feefeffe  
12:55:41.933 -> 3fffffa0:  feefeffe feefeffe 3ffee5d8 402026c0  
12:55:41.933 -> 3fffffb0:  feefeffe feefeffe 3ffe85d8 401011e1  
12:55:41.933 -> <<<stack<<<
12:55:41.933 -> 
12:55:41.933 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
12:55:41.967 -> 
12:55:41.967 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
12:55:41.967 -> 
12:55:41.967 -> load 0x4010f000, len 3460, room 16 
12:55:41.967 -> tail 4
12:55:41.967 -> chksum 0xcc
12:55:41.967 -> load 0x3fff20b8, len 40, room 4 
12:55:41.967 -> tail 4
12:55:41.967 -> chksum 0xc9
12:55:41.967 -> csum 0xc9
12:55:41.967 -> v00043510
12:55:41.967 -> ~ld
12:55:44.111 -> Param=0 value=a90  169
12:56:14.975 -> 
12:56:14.975 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
12:56:14.975 -> 
12:56:14.975 -> Exception (9):
12:56:14.975 -> epc1=0x43420e04 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000011 depc=0x00000000
12:56:15.008 -> 
12:56:15.008 -> >>>stack>>>
12:56:15.008 -> 
12:56:15.008 -> ctx: cont
12:56:15.008 -> sp: 3ffffdc0 end: 3fffffc0 offset: 0190
12:56:15.008 -> 3fffff50:  3ffee080 3ffee650 00000046 3ffee5ec  
12:56:15.008 -> 3fffff60:  3fffda00 3ffee650 00000046 402025a4  
12:56:15.008 -> 3fffff70:  40202c65 00000046 3ffee5ec 402010e2  
12:56:15.008 -> 3fffff80:  3fffdad0 00000000 000000cf 40201160  
12:56:15.008 -> 3fffff90:  00000100 00000000 feefeffe feefeffe  
12:56:15.042 -> 3fffffa0:  feefeffe feefeffe 3ffee5d8 402026c0  
12:56:15.042 -> 3fffffb0:  feefeffe feefeffe 3ffe85d8 401011e1  
12:56:15.042 -> <<<stack<<<
12:56:15.042 -> 
12:56:15.042 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
12:56:15.042 -> 
12:56:15.042 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
12:56:15.042 -> 
12:56:15.042 -> load 0x4010f000, len 3460, room 16 
12:56:15.075 -> tail 4
12:56:15.075 -> chksum 0xcc
12:56:15.075 -> load 0x3fff20b8, len 40, room 4 
12:56:15.075 -> tail 4
12:56:15.075 -> chksum 0xc9
12:56:15.075 -> csum 0xc9
12:56:15.075 -> v00043510
12:56:15.075 -> ~ld
12:56:17.208 -> Param=0 value=c40  196
12:56:36.567 -> 
12:56:36.567 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
12:56:36.567 -> 
[... 204 lines cut ...]
12:59:29.935 -> sp: 3ffffdc0 end: 3fffffc0 offset: 0190
12:59:29.935 -> 3fffff50:  3ffee080 3ffee650 00000046 3ffee5ec  
12:59:29.999 -> 3fffff60:  3fffda00 3ffee650 00000046 402025a4  
12:59:29.999 -> 3fffff70:  40202c65 00000046 3ffee5ec 402010e2  
12:59:29.999 -> 3fffff80:  3fffdad0 00000000 00000082 40201160  
12:59:29.999 -> 3fffff90:  00000100 00000000 feefeffe feefeffe  
12:59:29.999 -> 3fffffa0:  feefeffe feefeffe 3ffee5d8 402026c0  
12:59:29.999 -> 3fffffb0:  feefeffe feefeffe 3ffe85d8 401011e1  
12:59:29.999 -> <<<stack<<<
12:59:29.999 -> 
12:59:29.999 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
12:59:29.999 -> 
12:59:29.999 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
12:59:29.999 -> 
12:59:29.999 -> load 0x4010f000, len 3460, room 16 
12:59:29.999 -> tail 4
12:59:29.999 -> chksum 0xcc
12:59:29.999 -> load 0x3fff20b8, len 40, room 4 
12:59:30.051 -> tail 4
12:59:30.051 -> chksum 0xc9
12:59:30.051 -> csum 0xc9
12:59:30.051 -> v00043510
12:59:30.051 -> ~ld
12:59:32.147 -> Param=0 value=d60  214
```

geraldFS, 2021-11-05:
fumesTempAddr = 0x5A

It works for you with that value?

Yes!

philibertc, 2021-11-05:
Cool! One problem solved quickly
For the water pressure, if it is constantly 3× higher you can edit the [line 369](https://github.com/philibertc/micronova_controller/blob/ee899ee48db1be8ea5414a327f024a538e55d4d3/micronova_controller.ino#L369) by adding ` / 3`.

geraldFS, 2021-11-05:
I've already thought about this, but that shouldn't be the solution.

I am afraid I am using the wrong value or my oven is not delivering it.
I have to deactivate the water pressure measurement during the initial setup as soon as the pipeline set is installed.
Therefore I fear that the value in the form no longer exists.

Italian technology that inspires ;-))

But I can live very well without the water pressure, there are other things that press me more.

philibertc, 2021-11-05:
> I've already thought about this, but that shouldn't be the solution.

Yes, but only until then. There's still the flame power reading to fix, right?

geraldFS, 2021-11-05:
With the new .ino of today, errors are gone, now the heating power requirement goes.

the value that is displayed for water pressure also corresponds to what has been transferred, think either the oven delivers a fake value.

I just drained the water because I installed a small return increase and the value stayed exactly the same ...

closed 2021-11-05 by geraldFS

geraldFS, 2021-11-15:
Hi there,

I can confirm that the ITC Layma (18Kw) works with the Micronova Mod 0047 EVO.

There are two changes to the standard version:
- fumesTempAddr = 0x5A
- water pressure is not used by the control; it must always be deactivated. (see manual)

But otherwise TOP!

philibertc, 2021-11-15:
Hello,
thanks for the contribution!