philibertc/micronova_controller #22: Problems with power OFF / power demand

geraldFS, 2021-11-26:
Hi there,

QA strikes again ;-)

Since the penultimate change to the source code, I have had the symptoms that I cannot switch off the oven if it has been running for a long time.
I can shift gears, but it continues to burn stubbornly.

The water temperature and room temperature are updated and are valid.
Only the switch on / off and the power requirement are frozen.

When I switch off the oven on the display, everything is updated and works again.

Regards
Gerald

philibertc, 2021-11-26:
Hello, I will roll back to the previous commit, that new changes were
untested but I forgot to switch on dev branch.

philibertc, 2021-11-26:
I revert the changes

geraldFS, 2021-11-26:
Thanks!
Reacts again :-)

geraldFS, 2021-12-07:
Hi there,

with the version dated December 4th, 2021 - the same phenomenon occurs again!

Switching on is possible
Switching off is not possible

Values are all displayed accordingly.

philibertc, 2021-12-07:
Hello, that's strange...
I will inspect this soon.

philibertc, 2021-12-07:
Hello, I reread the code, but I do not see anything that could influence the extinction. I will have to do some investigation.

philibertc, 2021-12-12:
Hello, until I have more information about this problem, I will roll back to the previous functional version.

philibertc, 2022-02-27:
Hello,
I'm going to make some simplifications in the program this weekend (especially in the web settings that I'm going to minimize to the wifi configuration until I know how to manage the rest better). This should make it easier to diagnose bugs like this one!

Best Regards,
Philibert

philibertc, 2022-09-11:
Hello,
Can I close this issue?

Sincerely,
Philibert

geraldFS, 2022-09-20:
Yes!

closed 2022-09-20 by geraldFS