philibertc/micronova_controller #25: make it works with ESPHOME

raziel83257, 2021-12-22:
It would be wonderfull to make it works with ESPHOME, that is also integrated with HomeAssistant:
https://esphome.io/index.html

philibertc, 2021-12-22:
Hello, I was thinking something more like this: https://github.com/philibertc/micronova_controller/discussions/15 
For now I will make some other small improvements.
One day I'll probably make an esphome version.

Sincerely,
philibertc

geraldFS, 2022-09-20:
Perfect!
ESPhome would be the gold edge solution ;-)

Gerald

vyruz1986, 2022-11-10:
[This project](https://github.com/edenhaus/esphome-extraflame) offers a ESPHome component, pretty should it should work with the hardware of this project, as both software solutions (the arduino file in this repo, and the ESPHome config of the other repo) configure RX/TX pins which go to the stove's data pin