philibertc/micronova_controller #28: Parameter based on model/type?

liquidacid, 2021-12-29:
Hi,

Currently im using your tool to fill my Home Assistant. It seems there is not really a good database of addresses available. I have an Artel Hydro stove and would like to share my address list based on the board in it. Maybe it would be awesome to build up a database of addresses?

Im about to adding more stuff which is interesting like fan speed, settings like amount of added pallets per time slot etc so you can combined with water temperature see if your stove needs to be cleaned over time etc.

philibertc, 2021-12-29:
Hello
>It seems there is not really a good database of addresses available

Yes, I'm building it with time, I'll soon create a document summarizing various things about the protocol used and the addresses.

>Im about to adding more stuff which is interesting like fan speed, settings like amount of added pallets per time slot

Unfortunately these parameters have to be written to the EEPROM, which is a memory that wears out rather quickly (about 10000 writes) and I haven't yet found a safe way to limit the number of writes in case of a bug.

Sincerely, philibertc

liquidacid, 2021-12-29:
I do understand, i don't want to write the values but only read them to make sure what the current settings are. They only should be sent once to the MQTT topic once they are changed in the stove i think so you can perfectly see the details.

I also got a very different set of addresses.

I suggest to make some different species in a seperate file where you can choose from to select the right stove?

philibertc, 2021-12-29:
> but only read them

Okay, in this case it's easy, I just didn't take the time to test the EEPROM reading, but it won't take me long to set up.

> I suggest to make some different species in a seperate file where you can choose from to select the right stove?

Yeah, that's a good idea, I will do that quickly.

liquidacid, 2021-12-29:
Im not asking you to do this, i can help if you like but if i only make it for myself its probably not the cleanest way of coding it.

philibertc, 2022-01-02:
Hello
Don't worry, I have to do it. If you want to make tests you can try to modify the program: https://github.com/philibertc/micronova_controller/blob/master/tests/esp8266/test2_read.ino
Sincerely, philibertc