philibertc/micronova_controller #32: problem with micronova card of the year 2010

riccmike, 2022-02-07:
Thanks for your work, I tried to apply it to my Pellets Ecoteck stove with micronova board of the year 2010. With your code I didn't have any messages in the Serial Monitor, so I modified the code to understand what happens. I found that in my case, in the checkStoveReply function the rxCount variable is always zero. How has this ever happened to you? Perhaps the firmware of my board does not support communication, even if the serial connector is present.
I thank you in advance.

philibertc, 2022-02-12:
Hello, if there is the connector it is almost sure that it can communicate, because it can be useful in some cases for the repairer in case of failure.
This may be due to a bad connection. You can try this program to see if the stove reacts: https://github.com/philibertc/micronova_controller/blob/master/tests/esp8266/test1_on_off.ino
This program simply turns the stove on and off two seconds later.

Sorry for my late reply,
Sincerely,
philibertc

philibertc, 2022-02-16:
Hello,
This is really embarrassing.
Are you sure you have connected the cables correctly?
[image]
[image]
Sincerely,
philibertc

riccmike, 2022-02-16:
next weekend, I take the stove out of its position and retest the connector. Let's hope for my mistake. I will keep you updated.