philibertc/micronova_controller #34: Add compatibility for ESP-01 and Breakout board

wes1993, 2022-02-20:
Hello, 
could you help me to add compatibility for this board?
https://it.aliexpress.com/item/1005001970616935.html?gatewayAdapt=glo2ita&aff_fcid=56c564640e8546c3b341b78030ca809e-1645366026748-07392-_AOI1W4&tt=CPS_NORMAL&aff_fsk=_AOI1W4&aff_platform=shareComponent-detail&sk=_AOI1W4&aff_trace_key=56c564640e8546c3b341b78030ca809e-1645366026748-07392-_AOI1W4&terminal_id=ca5e0087d8a249f4b348a3734cd59a20&afSmartRedirect=n

The breakout board can convert from 5V to 3.3V and also adapt the signal but need TX and RX pin, in this board we have only one serial PIN so i don't know how can i adapt this...

Best Regards
Stefano

philibertc, 2022-02-21:
Hello, 

It will not be possible to adapt this breakout board because it is not advanced enough, it would need a few more GPIOs.
However, it should be possible to adapt the program for the ESP-01, still using the circuit with the optocouplers, but sacrificing some features like the serial port, which is very useful if everything does not work as expected.

Best Regards,
Philibert

wes1993, 2022-02-21:
Thanks a lor Philibert for your fast reply :-D
So i think i will keep the ESP8266 i just need to know what are the optocouplers to buy, do you have some link?
And did you know name/model of the female connector of the serial port on the micronova side?

Thanks a lot
Stefano

philibertc, 2022-02-21:
The optocouplers are PC817, 3 are needed
The esp8266 that fits best is the wemos d1 mini, which you can find by searching "d1 mini" on AliExpress or Amazon.
For the cable, this should be adequate: https://it.aliexpress.com/item/1005003638697583.html with 4 pins, 50cm mini and a female connector on one side.

Best Regards,
Philibert

wes1993, 2022-02-21:
Thanks again for your fast reply!!! :-D
I have found one Wemos D1 mini :-D so perfect i have the esp8266 i kust need the PC817... I have found two type of PC817 could you help me plese?:
This:
https://it.aliexpress.com/item/32719957788.html?gatewayAdapt=glo2ita&_randl_currency=EUR&_randl_shipto=IT&src=google&src=google&albch=shopping&acnt=494-037-6276&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=12556513086&albag=120214549340&trgt=1481207949271&crea=it32719957788&netw=u&device=c&albpg=1481207949271&albpd=it32719957788&gclid=Cj0KCQiAjc2QBhDgARIsAMc3SqSGqhPlUrNLW7T7jHOnzI231ygWAiqgvWu7u5Sa2mB74u7F-b-nZsIaAtddEALw_wcB&gclsrc=aw.ds&aff_fcid=c00fb8360b8a443daf1612cbae51c7a1-1645445315898-01528-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=c00fb8360b8a443daf1612cbae51c7a1-1645445315898-01528-UneMJZVf&terminal_id=ca5e0087d8a249f4b348a3734cd59a20&afSmartRedirect=n

or this:

https://www.ebay.it/itm/173890434826?chn=ps&norover=1&mkevt=1&mkrid=724-128315-5854-1&mkcid=2&itemid=173890434826&targetid=1339125098357&device=c&mktype=pla&googleloc=1008736&poi=&campaignid=9561703854&mkgroupid=128145120688&rlsatarget=pla-1339125098357&abcId=1145978&merchantid=6772416&gclid=Cj0KCQiAjc2QBhDgARIsAMc3SqToL_CQvP30uZ-8NPCU6kZrIMddK29z-jH5bDDqWCEQsD8umizm3BoaAnkiEALw_wcB

Best Regards
Stefano

philibertc, 2022-02-21:
The second one will be perfect!

wes1993, 2022-02-21:
Perfect thanks a lot!! :)
I’ll keep you updated and If I have some time I want to port this project on Blynk so we have an App and also HA integration

wes1993, 2022-02-22:
Hello Again @philibertc,
I have created the circuit and everything works fine!!!
I have only one question, there is a way to power on the ESP via 5V on the motherboard?

Best Regards
Stefano

philibertc, 2022-02-22:
Hello,
Glad everything is working!
I don't know if the stove is able to provide enough power via the 5V line, so I recommend using the 20V line with a converter like an LM2596 to get that down to 3.3V.

Best Regards,
Philibert

[image]

philibertc, 2022-02-22:
By the way, what is the model of your stove? So I can add it to the list of known models.

wes1993, 2022-02-22:
My stove is an El.Fire Venice Slim :)
But I should connect the GND together with the GND of ESP so did you think i could do that?

+ 20 V —> Converter —> Vin5V
GND    —> Converter —> GND of ESP

is the scheme correct?

wes1993, 2022-02-22:
P.S. I’m also trying to implement OTA updates, I will keep you update :)

Best Regard
Stefano

philibertc, 2022-02-22:
20 V —> Converter —> Vin5V *or 3V3 depending on the converter setting* but it is basically the same thing
GND —> Converter —> GND of ESP *= GND → Converter and GND → GND of ESP*

> My stove is an El.Fire Venice Slim :)

Thanks!
> I’m also trying to implement OTA updates, I will keep you update :)

That would be great, thank you very much!

wes1993, 2022-02-22:
Perfect thanks a lot :)

wes1993, 2022-02-22:
Hey another question.. sorry for the disturb :(
Did you know if there is a way to set the temperature and the clock?

philibertc, 2022-02-23:
Hello,
We are currently discussing it here: https://github.com/philibertc/micronova_controller/discussions/30 but for the moment the most reliable solution is to emulate the IR remote control.

Best Regards,
Philibert

wes1993, 2022-02-23:
ok perfect thanks a lot

wes1993, 2022-02-27:
Hello Philibert,
I have a question, I really really don’t understand why we need the 500 ohm resistor… could you help me?

Because I want, i case something won’t work, to attach the Micronova board to an arduino to convert TTL
To usb, but I think arduino is 5 v so I need to know if I just need to power the optocoupler via arduino and I have 5V or I also need to remove the resistor.

thanks a lot
Stefano

philibertc, 2022-02-27:
Hello,
an optocoupler works like this:
[image]
When we turn on the LED (on the left) the phototransistor on the other side becomes conductive. The 500 Ohms resistor (470 on the diagram) just allows not to burn out the led.

Electronically speaking, replacing the ESP by the Arduino works: the 500 Ohm resistor is also sufficient for 5V and not only 3V3. However, the Micronova board uses a strange serial configuration which is not available in the SoftwareSerial library for Arduino:
[image]
[image]
I will send you the procedure to use the Arduino as a serial converter, without using any program on it (just connecting to the RX and TX pins) and using a software like [YAT](https://sourceforge.net/projects/y-a-terminal/).

Best Regards,
Philibert

PS: I closed the issue but we can still discuss here.

wes1993, 2022-02-27:
@philibertc thanks a lot for your reply!! :) I will wait your instructions!! :)

philibertc, 2022-02-27:
I forgot a detail, in fact I realize that it is not possible to use to Arduino, even using it as a FTDI for example. The only solution is to use a program for the esp, which would play the role of a kind of bridge between the computer and the micronova board. I'm looking into how to do that.

wes1993, 2022-02-27:
Ok perfect thanks a lot!’ I will wait!’ :)

wes1993, 2022-02-28:
This is a WP of the mobile App:

[image]

what do you think? Do you like this?

bernardstef, 2022-02-28:
Hi !
I've been following this project and using it for a while ! Awesome to see a mobile app, I have the micronova wifi module (easy connect plus) + play store app and it's not the most reliable ..

wes1993, 2022-02-28:
I’m working on this, for now seems very powerful I just need to know how can we write value to eeprom in a safe way, I need the help of @philibertc for this and we could have the exact same function of the stock App.
This could be used for example for set the timer

[image]

best regards
Stefano

wes1993, 2022-03-01:
Hello,
I have some news :-D I have seen that someone has added the OTA code i will share my Blynk code also and i have added also the DUMP memory via a simple terminal command without connecting the device to PC.

@philibertc could you tell me if the values of params are decimals or HEX values?

Because for now i need to write the value in decimal otherwise the terminal can't accept the value.

Best regards
Stefano

wes1993, 2022-03-01:
Added also some Graph
[image]

philibertc, 2023-02-10:
Hello,
I'm reopening this issue as I seem to have missed a lot of the discussion, sorry for that.
Has there been any progress?