philibertc/micronova_controller #42: Extraflame luisella stove not working/responding, other 4-pin pinout?

jeanpolle, 2022-09-26:
Hi everyone!

I have a luisella (extraflame) stove with 4-pin connector on on the back, but i can't get it working with the esp...
Before i take the stove off the pipe to have a look at the card internally; i thought i'd ask here; maybe i'm missing something obvious, i'm rather new to arduino ;)

i got the PCB from tindie, and have an ESP8266 wemos D1 mini fitted. Soldering all components went smooth. don't think i made an error there. (pics attached)

Programming the ESP doesn't work when it is fitted tot the PCB board however... it has trouble connecting, and the led starts blinkin erratically.
Once I remove it from the pcb, uploading the code works just fine. Can anyone comfirm the same behaviour?
i use the micro usb port to power the ESP.

Wifi setup works fine, I have a broker set up (mosquitto in a docker container) and can see the "pong" appear nicely every 25s.

I checked the serial port on the stove before connecting (good thing i did ;)), and when measuring the voltages i noticed a difference in pinout,  I found it to be: (gnd - 5v - 20v - ser) instead of  the documented (gnd - ser - 20v - 5v)

I hooked up the board, but nothing comes up other than the pong

-I tried the 2sec on/off program, but no reaction from the stove
-Test read.ino with a serial monitor running doens't show anything messages.

anything else you can think of? anyone found serial messages to be turned off in some admin menu-thing on the stove itself?

thanks!

J

[image]
[image]
[image]
[image]
[image]

philibertc, 2022-09-26:
Hello!
First of all, thank you for detailing your issue.
> don't think i made an error there.

Yes, it looks all good.
> Once I remove it from the pcb, uploading the code works just fine. Can anyone comfirm the same behaviour?

I have the same behavior, it's because of the link between D0 and RST which is used to add the possibility of deep sleep.
> i noticed a difference in pinout, I found it to be: (gnd - 5v - 20v - ser)

The serial and 5V pin are both at 5V, how did you tell the difference?

I hope I have answered a majority of your questions!
Do you have some resistors (above 1kOhm) and a multimeter at your disposal? There is a way to surely differentiate between serial and 5V.

Have a great day,
Philibert

jeanpolle, 2022-09-26:
Hi Philibert,

First of all, thanks for the excellent work done so far, great stuff! quite stunned :)

> The serial and 5V pin are both at 5V, how did you tell the difference?

Well..I wasn't sure what voltage to expect on the serial pin when measuring.
I have a pin which caries no voltage referenced to ground, and since I already had a 5v and 20v present; i figured it was the serial pin .. (was gonna ask how to be sure of this ;))

So that would explain why i get nothing; the 5v pin i measure will be the serial as per original pinout...
I do have resistors/multimeter at my disposal so if you got a trick to make sure, i'm all ears

Now what to do with the pin with no voltage... i measured it a few time, and to the ground pin i have 0v where you have 5v :)

i'll hook the pinout up as per spec/doc tonight or tomorow

Thanks for the help!

J

philibertc, 2022-09-26:
> I do have resistors/multimeter at my disposal so if you got a trick to make sure

Okay, nice!
To check which pin is serial, put your multimeter between GND and one of the two 5V pins. After, connect this pin to the GND via a resistor of more than 150 Ohm (not to make a short circuit or exceed the limit of 0.25W of a resistor), you can check if this pin is now at 0V. In this case, it is the serial pin, otherwise, it is the power supply pin.
Is this clear? I could make a diagram if you want.

From what you have described, the pinout described on the doc should actually be good, but it is always good to check.

Regards,
Philibert

jeanpolle, 2022-09-28:
Hi Philibert

little time, little knowledge, so progress is slow :)

I measured all pins again, and tried the resistor test to pull the serial to 0V. 
It identified the documented 5V pin as the serial, good news! 
The other pin (power) however remains at 0v instead of 5V all the time.

nevertheless, I corrected the wiring to the PCB and hooked it up to the stove but got no other messages from the ESP on the subscribed broker channels than "connected" on /pong.

nothing on /ambtemp, /state or /onoff

I connected another mqtt client to the broker and published messages on these topics, to check if my broker setup was working correctly. They showed up just fine on the first client. 

I can however turn on the stove by publishing ON to /intopic, but the stove wont turn off when sending OFF.

I then loaded the on/off test.ino, and that worked. The stove went on and off again. Nice! 
So why wont it turn OFF then when publishing OFF to /intopic.... weird. 

it looks like a problem in this earlier thread...
https://github.com/philibertc/micronova_controller/issues/31

any ideas?

thanks!

J

philibertc, 2022-10-05:
Hello, first of all, sorry for my late reply
> So why wont it turn OFF then when publishing OFF to /intopic.... weird.

It is indeed strange. The fact that the ON/OFF test works already lets us know that it is not an address problem. When you send the message to turn off, does the blue LED of the WeMos blink (once and then several times right after)?

> it looks like a problem in this earlier thread...
https://github.com/philibertc/micronova_controller/issues/31

The problem does seem to be the same. I'll try to build a more advanced test program to find out what's wrong. The OFF command is normally built the same way as the ON, if one works I wonder what is blocking the other.

Bests,
Philibert

jeanpolle, 2022-10-06:
> Hello, first of all, sorry for my late reply

No problem, enough work to do here :)

> It is indeed strange. The fact that the ON/OFF test works already lets us know that it is not an address problem. When you send the message to turn off, does the blue LED of the WeMos blink (once and then several times right after)?

I remember the led blinking once when sending ON, and not blinking when sending OFF.
i'll try to confirm this later tonight.

> I'll try to build a more advanced test program to find out what's wrong. The OFF command is normally built the same way as the ON, if one works I wonder what is blocking the other.

the fact that no other parameters are published on any of the other topics (ambtemp / status / ...), would that solely be due to different addresses? 

Thanks!

J

jeanpolle, 2022-10-25:
Hi guys,

got a bit of time on my hands; can anyone point me in a direction of things i can try? 
Verify my method on how to run a serial monitor? 
(running one in arduino IDE with the ESP connected via microUSB, and the serial connections to the stove, but i get absolutely nothing :))

J

philibertc, 2022-10-27:
> EDIT:
> The led blinks 1x short + 4x short when sending ON
> The led blinks 4x short when sending OFF

Hello!
All is good for the ON command, but it seems that the OFF command is not sent because the LED does not flash once in the first place.

Can you try reversing the ON and OFF command to see if the stove turns off (sending "on", since it will be reversed).

Bests,
Philibert

jeanpolle, 2022-10-27:
Hi Philibert,

Can you be more specific on how i should reverse the commands?

Would this mean altering the INO file and reuploading it to the ESP? 
(as in; would reversing values in line 43 and 44 suffice? i'm a total noob at this)

Or am i missing an easier way?

kind regards!

J

philibertc, 2022-10-27:
that modification bypasses the two if conditions using ```if (true)```

jeanpolle, 2022-10-27:
Thanks so much; that works!

So the trouble lies in the controller reading the stove's status to be able to fullfill the condition, and only then send the command. 

I don't get any data on other topics, so that problem is related....

What are the next steps to determine what goes wrong? other addresses, and how to find them?

Thanks for taking the time to help me out here; i'm learning a bunch of stuff ;)

Jan

msaitz, 2022-11-22:
I'm having the same issue with a Laminox Idro marichiara, I can only turn it ON with no additional actions or readings from the stove.
I'll be doing some additional testing to see if my issue fully matches @jeanpolle

jeanpolle, 2022-12-17:
did anyone make any progress?

@msaitz did you manage to get readouts?

msaitz, 2022-12-23:
@jeanpolle I haven't had time for it yet!