philibertc/micronova_controller #44: MCZ Ego

ThomasTom86, 2022-10-23:
Good morning,
 I came across your amount when I was looking to make my pellet stove smart.  Unfortunately, my oven (MCZ Ego Air 2017) does not have a CN 13 port installed.  Is it still somehow possible to integrate this model into the iobroker via mqtt?  There is an original Wifi module for this oven (unfortunately very expensive).  Thanks

[image]

[image]

jeanpolle, 2022-10-25:
If i had to guess; i'd say the 4 pins marked "com" are candidate for serial communication....

philibertc, 2022-10-27:
Hello, 
According to the video, the "COM" pins seem to match. The use of only three cables also seems to confirm that it is the same protocol. You can make measurements with your multimeter by connecting the GND of the voltmeter to the 2nd from the left, then measuring the rest of the pins with the red probe.

Best Regards,
Philibert

ThomasTom86, 2022-10-27:
Thank you for your message.  This is the measurement result from the Com port.

Best regards
Tom
[image]

ThomasTom86, 2022-10-28:
I still found that
[image]

ThomasTom86, 2022-10-28:
[image]

closed 2023-09-10 by ThomasTom86