philibertc/micronova_controller #47: Wiring for La Nordica Extraflame Emma Plus

vyruz1986, 2022-11-12:
Hi,

First of all, thanks for putting this project together!

I'm trying to use this project to control/monitor the aforementioned pellet stove, but am struggling to find the right connections.
The control board I have is the one listed in [this PDF](https://www.extraflame.it/support/index.php?option=com_docman&Itemid=26&task=doc_download&gid=916) on page 40, 'board code' (whatever that may mean) 00227575.
Here's a picture of the board (sorry for poor quality, I took it from a webshop since mine is in the stove and I don't want to spend time disassembling it): 
[image]

As you can see the 'Seriale' connector is positioned otherwise than in the readme of this repository.
In my stove, there is also a cable connected to it (tested all 3 wires, it's just an extension cable), which has a panel mount connector (exactly the same type as on the board) on the rear of the stove:
[image]

I also noticed that the extension cable only has 3 wires, while the board connector has 4 pins.
I measured all 4 pins with my DMM, and got the following results (numbering following your drawing of the connector in the readme):
1. GND, blue wire in extension cord (confirmed by measuring continuity between this pin and the negative CMOS battery pin
2. Black wire in extension cord, measuring constant +5V
3. Red wire in extension cord, measuring fluctuating +17-19V (approx every second) 
4. No wire in extnesion cord, measuing constant +5V

I also measured all pins with my oscilloscope, but no obvious digital signal is visible.

So I'm at a loss as to which pin is the serial data pin, can someone help me out?

jeanpolle, 2022-11-16:
Hi!

looks like you are as confused as i was/am.... and are possibly running into the same problems. :)

have a look here; 

https://github.com/philibertc/micronova_controller/issues/42#issue-1385697452

i found i can send stuff to the stove on the pinout philibert descibes (i assume yours will be the same) but cannot get a readout from the stove so we had to adapt the ino a bit to get the messages sent.

haven't scoped the output of my stove yet..

J

vyruz1986, 2022-11-16:
I forgot to mention I was (beside this and [this](https://github.com/netdata-be/PalletHeater-REST-API/issues/1) repository) also looking into [this](https://github.com/edenhaus/esphome-extraflame) solution, which I found the most attractive since it uses ESPHome allowing me to easily modify and upload new firmware in the future. With some help from @edenhaus I was able to get it working yesterday. The serial pin was indeed the 5V pin (pin #2) which was brought to the backpanel connector via the black wire.
I used this schematic to hook up my ESP8266: 
[image]

It's different from the one described here , which I believe is better (since it uses optocouplers and offers complete isolation of ESP and stove), but as discussed above I couldn't get this schematic working :(
Also note that @edenhaus's solution doesn't require the `Enable_RX` pin which this solution needs.

edenhaus, 2022-11-17:
@vyruz1986 I have a logic shifter between the esp and the oven as the esp is not 5v tolerant.
I used a chinese copy of https://www.sparkfun.com/products/12009

vyruz1986, 2022-11-17:
> @vyruz1986 I have a logic shifter between the esp and the oven as the esp is not 5v tolerant. I used a chinese copy of https://www.sparkfun.com/products/12009

I know but I didn't have any logic shifter available, so I started with ESP8266 which is 5V tolerant (although not officially, I've used this MCU with 5V logic many times without problems. Also see [this](https://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/)).
I have ordered logic shifters and will switch to ESP32 as soon as they arrive.

edenhaus, 2022-11-17:
Nice to know about the 5v tolerant. Maybe I will change on my load weight cell project to 5V to have a better precision. Thanks :)

@vyruz1986 It would be great to add the schematic above with the logic shifter to my repo. As you have already drawn it, feel free to add it with a PR :)

philibertc, 2022-11-17:
Hello,
> I also measured all pins with my oscilloscope, but no obvious digital signal is visible

This is normal: the stove does not send any information until it is asked.

> So I'm at a loss as to which pin is the serial data pin

According to this diagram it must be the second pin "Black wire in extension cord, measuring constant +5V" you measured:
[image]
the position of the "seriale" connector varies between the different models of micronova cards.

Best Regards,
philibertc

philibertc, 2022-11-17:
Have you tried any of the test programs?

bachtsevanis, 2023-02-20:
@vyruz1986 can you explaine how you connected with the board? I don't know how to read the diagram that you upload. My stove board has the same serial as yours. It's a Micronova J034_7

vyruz1986, 2023-02-21:
> @vyruz1986 can you explain how you connected with the board? I don't know how to read the diagram that you uploaded. My stove board has the same serial as yours. It's a Micronova J034_7

Read [this chapter](https://github.com/edenhaus/esphome-extraflame/blob/main/docs/connection.md#finding-the-right-pins) to find out how to identity the pins on your stove connector.
Once you found GND and DATA, connect GND to the ESP, and use a 1N4148 diode to wire the stove's DATA pin to the ESP's RX pin, with the black mark on the diode facing the ESP. Then branch another wire from the stove's DATA pin and connect it to the EPS's TX  pin.

bachtsevanis, 2023-02-22:
@vyruz1986 I suppose that you are not using external power ie USB power adapter like this solution?

rotragit, 2023-10-28:
Extraflame doen't wire the 5V pin on the backside socket. But you can get it from the socket on the controller. Use that one.

pippo93, 2024-11-10:
Any update about this issue? I have now finished the assembly of pcb as showed in this repo but I have the same Micronova Controller of the firat comment…