philibertc/micronova_controller #53: Question about 5V Pin

eni23, 2022-12-16:
Hey @philibertc 

I have a pellet stove (L'Asolana Marina) which clearly has a Micronova controller board, but i'm unable to access the PCB for interfacing with it. But it has a SUB9 connector on the back with has 3 connected pins (GND,20V,Serial).

 I would like your circuit, but the optocoupler that is connected to `RX_ENABLE` needs 5V which i do not have with the 3P-connection. Since its only driving a  optocoupler, i was thinking about using a step-down board and use the 20V to drive this optocoupler. Do you think this will work or is it too dangerous to try it? Any ideas?

The official micronova WIFI board [pcb front](https://www.micronovasrl.com/wp-content/uploads/2016/02/DSC_2024.jpg) [pcb back](https://www.micronovasrl.com/wp-content/uploads/2016/02/DSC_2020.jpg) also only connects 3 pins and seem to directly convert the 20V down as well.

Thanks in advance :)

philibertc, 2022-12-26:
Hello!
Using a buck converter to do this is not a problem, you can definitely use this method.

Have a great day,
Philibert

eni23, 2022-12-27:
Thanks a lot for your answer. I tried it today and the stove replied and turned on as excepted. I will close here but will create a MR in the next few days to add my model to the supported ones in README.md. 

Have a nice day as well :)

closed 2022-12-27 by eni23