philibertc/micronova_controller #58: Safer way to turn the stove off

eni23, 2022-12-30:
Turning the stove on works fine over writing `0x01` to `RAM 0x21`, turning it off (sending `0x06` to `RAM 0x21`) seems to be problematic. My stove enters "turning off mode" but takes pretty long to do so (~50min), so this is not the procedure used by programs or the power button which takes around 10min. Searching around in the internet told me that i am not the only one which has this issue. The stove also has a infrared control, which seem to write commands to `RAM 0x58`. So sending 10 times `0x5a` to `RAM 0x58` with a delay of `100ms` turns the stove off safely.

You should consider changing your code to this method as well, since it seems the better way and also should work for all stoves.

There are other "IR commands" that can be written to `0x58` with a delay of `100ms`. To set power and temp, send it at least twice, for the power button (longpress) at least 10 times.

| State | Description   |
|-------|---------------|
| 0x54  | Power +       |
| 0x50  | Power -,      |
| 0x52  | Temperature + |
| 0x58  | Temperature - |
| 0x5a  | Power ON/OFF  |

philibertc, 2023-01-22:
Hello,
Thank you for your contribution!
I'm testing this out as soon as I can, then I'll probably incorporate it.
Bests,
philibertc

maxcalavera81, 2023-01-29:
I think I have the same problem. Because whenever I turn off the stove, I can turn it back on again. I don't really know what's going on.

paciotti, 2023-12-31:
Hello
I want to try this method for turning the stove ON and OFF since the original one is not working for my hydro stove.
How should we manage the checksum? If I'm not wrong the command should be {0x80, 0x58, 0x5a, 0x132} but the last value exceed the char dimension... Should we just use 0x32 instead?
Thank you for your clarification.