philibertc/micronova_controller #59: Arduino ide deepSleepInstant error

maxcalavera81, 2023-01-26:
Hello, I'm having a "deepSleepInstant" error and I don't know how to solve it. Does anyone have any idea how or what I'm doing wrong?

[image]

philibertc, 2023-01-26:
Hello, it seems that you have selected the esp32 version of the D1 mini, this program is made especially for the esp8266 version

maxcalavera81, 2023-01-27:
@philibertc thank you very much, problem solved. It really was the poorly selected board. Thanks a lot for the project.

philibertc, 2023-01-27:
Hello, I am glad to hear that. Could you please tell me the model of your stove if everything works well? You can make a pull request for this, and be listed among the contributors.

closed 2023-01-27 by philibertc

maxcalavera81, 2023-01-27:
My stuva is a cadel crystal .  I'll do some more testing and then I'll give you feedback.