philibertc/micronova_controller #60: States missing micro nova n100

maxcalavera81, 2023-01-28:
Hello @philibertc , I was here doing some tests and I noticed that there are missing states or possibly different states in your firmware. In the case I had a failure in the ignition and it continues to report loading pellets. Can you give me tips on how to solve this problem? My salamander is a crystal cadel with n100 board.

[image]

The machine shows a failure to ignition, alarm 5

Thank you very much for your attention

closed 2023-01-29 by maxcalavera81

philibertc, 2023-02-10:
Hello,
everything is working properly finally ?

Bests,
philibertc

maxcalavera81, 2023-02-10:
Hello @philibertc Thank you for your concern.  Yes, it's already working.  The problem was really with wemos.  There are small updates here that maybe could improve the experience with the salamander via Wi-Fi (I don't know if it will be possible or even easy to implement 😜😜).  One of them would be the online and offline status in mqtt because even if it is not online it always shows the online status.  The other thing I noticed these days is that when an alarm occurs, it is not possible to turn it off remotely because the switch immediately goes off.  Again, I don't know if such a change will be possible.  Smoke temperature is not displayed with real values ​​(I don't know if it only happens to me).  And as for the warnings, I had a lack of pellets in the stove when in reality it was a misfire.  I don't know what the behavior is in this case, but the stove's LCD shows the right alert.  These are small adjustments that could be improved over time.  I can't stop thanking you for the fantastic work you've done.  It was just what I needed to control my heat.  PS: if you add a web interface for those who don't have a home assistant, I think it would be really great...