philibertc/micronova_controller #62: How to make it work with Alexa

jayjay13011, 2023-02-11:
Hello philibert. 
All works so fine with my Extraflame Luisella. Thanks to you. But how I could make it works with Alexa. I try with fauxmo code but I’m a novice with all of this. 
Could you help me?

Thanks by advance!

philibertc, 2023-02-11:
Hello,
unfortunately I don't know very well how to integrate devices in Alexa. I will look at fauxmoESP to see how it works.
What home automation platform are you currently using?

Best Regards,
Philibert

PS: I leave this issue open, as a feature request or suggestion.

jayjay13011, 2023-02-11:
I’m using Home Assistant.

I tried to integrate fauxmo, but my code is not correct and all the exemples I find on forums are for relays or lights. So it’s difficult for me to adapt it.

And for information, do you know why I have to restart my D1 MINI regularly since I flash your code.
￼

As you can see, I switched off near 12:30 but I don’t have datas since 11:45 and unfortunately, datas stay as they were when the D1 mini stopped.

Thanks.

jayjay13011, 2023-02-20:
@silkyclouds Thanks for the tip. I tried but I have a problem with https and duckdns: I can't access locally to HA when I activate https. I'll try again this week-end.
@philibertc My problem in #3 is always there. In fact, I think the ESP8266 goes in deep sleep mode and never wakes up. I connected D0 and RST. Do you know what can be the problem? Except this, all is ok, but I no longer can use it due to this problem because I can't switch on it after one hour or two running.

Thanks.

philibertc, 2023-02-20:
Hello, you can try to comment or delete lines 525 to 529 (from if to })

jayjay13011, 2023-02-21:
@philibertc  I did it, but same result... :-(

philibertc, 2023-02-21:
Argh, can you see what's going on with a serial monitor at that moment?

jayjay13011, 2023-02-26:
Here is the result after freeze with Exception Decoder :

`Exception 0: Illegal instruction
PC: 0x4021a750
EXCVADDR: 0x00000000

Decoding stack results
0x401011c6: check_poison_block(umm_block*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc/umm_poison.c line 86
0x40100c60: interrupt_handler(void*, void*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 167
0x4010122c: check_poison_neighbors(umm_heap_context_t*, uint16_t) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc/umm_local.c line 59
0x40100b9c: interrupt_handler(void*, void*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 138
0x40224b24: mem_free at core/mem.c line 236
0x40100c60: interrupt_handler(void*, void*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 167
0x4010091c: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x40100b9c: interrupt_handler(void*, void*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring_digital.cpp line 138
0x4010091c: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x40216be6: loop_task(ETSEvent*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 273
0x401000ab: app_entry_redefinable() at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 386
0x401011c6: check_poison_block(umm_block*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc/umm_poison.c line 86
0x401011c6: check_poison_block(umm_block*) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc/umm_poison.c line 86
0x40101420: umm_malloc_core(umm_heap_context_t*, size_t) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc/umm_local.c line 47
0x401010f8: get_poisoned(void*, size_t) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc/umm_poison.c line 121
0x40211740: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::addDomainCacheItem(void const*, bool, unsigned short) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp line 2173
0x40101750: umm_malloc(size_t) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 912
0x40211740: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::addDomainCacheItem(void const*, bool, unsigned short) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp line 2173
0x401017c0: umm_poison_malloc(size_t) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\umm_malloc/umm_poison.c line 169
0x40216a5b: operator new(unsigned int) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\abi.cpp line 41
0x40212335: esp8266::MDNSImplementation::MDNSResponder::_udpAppendBuffer(unsigned char const*, unsigned int) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1163
0x4021263a: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSRRDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRDomain const&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1395
0x402127f4: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService const&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1502
0x4010091c: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x4010091c: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Jeremy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_main.cpp line 238
0x4021d5d9: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 174
0x40222354: _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 662
0x40221f04: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
0x40221f04: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
0x40221f04: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
[... cut ...]
0x40201878: getAmbTemp() at C:\Users\Jeremy\Documents\Arduino\PHILIBERT_ORIGINAL_MQTT_OK/PHILIBERT_ORIGINAL_MQTT_OK.ino line 400
0x402018d9: getStates() at C:\Users\Jeremy\Documents\Arduino\PHILIBERT_ORIGINAL_MQTT_OK/PHILIBERT_ORIGINAL_MQTT_OK.ino line 474
0x40201962: loop() at C:\Users\Jeremy\Documents\Arduino\PHILIBERT_ORIGINAL_MQTT_OK/PHILIBERT_ORIGINAL_MQTT_OK.ino line 523
`

After this, I have 2 other exception error. If you want, I can post them or send it to you in private.

Good week-end!