philibertc/micronova_controller #68: Only on status works

DEDEit, 2023-03-21:
Hello everyone, I managed to configure everything but unfortunately the only working command is the ignition.
The shutdown does not work and I do not receive data on temperatures, flames, etc.
How can I check?
[image]
[image]
[image]
[image]
[image]

Maxx2828, 2023-12-23:
Same for me. Any solutions?

[image]

goldrake1981, 2024-09-16:
The same for me. Only state (late) and room temperature works. I stay nothing works. I have an extraflame viviana and should have micronova m011_1

Bougnat63, 2024-11-24:
Hello, were you able to find a solution? I have the same card and I'm looking to make this little module

goldrake1981, 2024-11-24:
nothing. I'm sorry.

maxapple71, 2024-12-07:
I have the same card on a MCZ polar and I had the same problem, but I SOLVED!! I debugged the levels and found that the RX_PIN D3 pin of the D1 Mini never went down, not even grounding the pin and therefore could not receive anything from the control unit. I modified the circuit board using the D1 pin instead of D3 with a 10k pull up resistance and of course I assigned RX_PIN to D1 in the code and now it works! I hope it's the solution for you too, but check it out quickly

dackerman38, 2025-01-29:
@maxapple71  Hi, Can you share a picture or diagram ? Thanks

grutttepier, 2025-09-17:
Here @Zenomathics84 made a diagram for it.
[https://github.com/philibertc/micronova_controller/issues/31#issuecomment-2710487294](url)