philibertc/micronova_controller #69: unable to read values from the stove

hansf71, 2023-08-07:
Hello @philibertc , a short while ago I received your pcb. Installing the Wemos and the firmware was no problem, so far so good. Only I have a problem getting values from the stove. I've tried the test scripts and I am able to turn the stove ON and OFF with the onoff test script, but when trying to read one value or all values I never get any response. I added some extra serial output to verify the script is running, it is. But not one value is received. 

Your pcb is connected to the Micronova pcb and powered over usb from my laptop. The 3.3v power connection is not used. In the photo you can see the setup I am testing with. The stove is a Tim Sistem Aquas 15.

Do you have any suggestions what to do next to solve this problem? I am able to do any test you suggest. Do I need to check my pcb for a problem?

Hope to hear from you. Thanks!
Hans

[image]

esavalle, 2024-01-07:
Hello @hansf71,
Did you manage to solve your problem ? I managed to get my stove info using [eni23's repository](https://github.com/eni23/micronova-controller) but I would like to get the MQTT integration that is only present in philibertc's repo. 

I know that the communication with the stove is working thanks to eni23's work but I can get any answer from philibert's repo. So if you have managed to solve it, it would help me a lot !

hansf71, 2024-01-07:
Hi @esavalle,
I was not able to fix the problem yet. Due to lack of new input. But I still want to have a mqtt connection to my stove. Were you able to turn the stove on or off with philibertc's repo? I have an ESP32 somewhere and I will try to get eni32 repo working. From there I would think it would be possible to add mqtt support.
I will let you know. If you have any input or advise to get eni32 repo working, I'm all  ears!
Greetz
Hans

esavalle, 2024-01-08:
I indeed was already able to turn on and off the stove with philibertc's repo but as it was not working for the other command, I discovered eni23's repo which allow me to "decode" the communication. 
eni23 was able to guide me a lot on how to make it work after a [few back and forth with him](https://github.com/eni23/micronova-controller/issues/8). Unfortunately, there is no MQTT integration yet but this is nothing too complicated I guess. I'll try to merge the two into something that will be functional for me and hopefully for someone else.