philibertc/micronova_controller #80: Compatibilité poele MCZ flat confort air

herve8940, 2024-12-01:
Bonjour à tous,

Pourriez-vous me dire si mon poele mcz flat confort air est compatible avec le projet? Voici la reference de ma carte mere:41450907601.
D avance merci pour votre aide.

Hervé

philibertc, 2024-12-01:
Bonsoir,

Pouvez-vous m'indiquez si vous avez accès à un port série similaire à celui-ci (en dessous de Schematics) ? https://github.com/philibertc/micronova_controller/wiki/How-to-build-it#stoves-mainboard-pinout
Je ne parviens pas à voir s'il y en a un sur les photos disponibles sur Internet.

Merci

herve8940, 2024-12-01:
Bonsoir,

Voici la photo de ma carte mere:
[image]
Merci beaucoup pour votre aide.
Hervé

philibertc, 2024-12-01:
Malheureusement il ne semble pas y avoir le port nécessaire sur cette carte 😐

herve8940, 2024-12-01:
Mince! Je suis déçu 😞 voici le schéma électrique de la carte mere:
[image]

philibertc, 2024-12-01:
Désolé... Moi aussi, c'est la première fois que je tombe sur une carte mère sans port Série, je vais ajouter cette référence à la documentation pour plus tard
Le schéma électrique ne semble pas avoir été uploadé, c'est vrai que ça peut être intéressant d'y jeter un coup d'œil pour confirmer

herve8940, 2024-12-01:
Je suis quand meme étonné car je me dis qu il dois avoir une solution pour le dépannage. Il y a des connecteurs disponibles.
Voici le schéma:
[image]

herve8940, 2024-12-02:
Si ce n'est pas faisable avec votre projet, auriez-vous une autre méthode à me proposer ? 
merci pour votre aide.
Hervé

philibertc, 2024-12-02:
Bonjour,

Effectivement le schéma confirme qu'il n'y a pas le port nécessaire malheureusement.
Comme autres solutions je vois seulement rajouter un transistor sur le bouton de l'écran pour simuler un appui (mais aucun retour avec cette solution) ou alors essayer de comprendre comment le module boutons/écran communique avec la carte mère (ce qui est beaucoup plus délicat)

Philibert

PS: au vu de cette discussion, faut-il que je maintienne la commande sur Tindie ?

philibertc, 2024-12-07:
Bonjour,
C'est une télécommande IR ? Si oui, il est alors éventuellement possible de capturer les signaux transmis pour les simuler ensuite avec un arduino ou équivalent.

closed 2024-12-07 by philibertc