philibertc/micronova_controller #82: the stove does not turn on

maxapple71, 2024-12-10:
By sending the ignition command the stove turns on and loads pellets but it does not turn on and then there is an error. The glow plug is cold...If I turn on the stove from its display everything works correctly. It's a MCZ Polar hydro, communication with the stove works and the on/off test works correctly, can read all data...how can I do it?

maxapple71, 2024-12-11:
I paid attention and if I turn on the stove with the button I hear the tick of a relay (which powers the glow plug), while I don't hear it when I send the command {0x80, 0x21, 0x01, 0xA2}...
[just turned on with the power button.log](https://github.com/user-attachments/files/18097323/just.turned.on.with.the.power.button.log)
[OFF.log](https://github.com/user-attachments/files/18097324/OFF.log)

maxapple71, 2024-12-14:
Thanks to fabrizioromanelli I was able to turn on the stove properly! With this {0x80, 0x58, 0x5A, 0x132} my Mcz started!

c4coer, 2025-05-08:
bonjour comment avez vous trouvé ce code ?